atc1441/CH559sdccUSBHost #2: Makefile for Linux

maikmerten, 2020-01-15:
Hi there,

thanks for providing this very interesting project!

I created a Makefile to build the project on Linux. Is this something you'd consider for inclusion?

atc1441, 2020-01-15:
Hi. 
Yes i would test it on a mashine and then add it to the repo :)

maikmerten, 2020-01-15:
Great! I attached a Makefile that "works for me" and hope it works for you as well. I'm using sdcc as packaged by Ubuntu and everything compiles fines (albeit with warnings) but currently don't have hardware yet to test the output.
[Makefile.zip](https://github.com/atc1441/CH559sdccUSBHost/files/4066738/Makefile.zip)

atc1441, 2020-01-16:
Thank you for the makefile, i tested it.
could not get it to work right now, it says no aims and no "make" controllfile was found. 

i dont make very well, how did you compile it? 

As for the Licence. the CH559.h file is edited to work with the SDCC enviroment and comes partly from this original example zip file: http://www.wch.cn/downloads/CH559EVT_ZIP.html

maikmerten, 2020-01-16:
Oh, you need to place the file "Makefile" in the same directory as the "compile.bat" file and simply invoke "make" in the directory, without parameters. By default the "all"-target should be executed.

You can try
"make" or "make all" - this should build the source files
"make clean" - this should remove the files created by the build process

I find that sdcc 3.5.0 (distributed with Ubuntu 18.04) cannot compile the source code, but sdcc 3.8.0 (distributed with more recent Ubuntu versions) works.

As for the license: It's a bit sad that the code examples don't appear to have any licensing information, limiting its usefulness. I see that you created a GitHub issue with the maintainers of that code.

atc1441, 2020-01-16:
Did everything allready as you wrote now, so these are still my error messages.

Maybe i have the wrong make, using kali linux and sdcc 3.8.0

The issue i created was meant for the usb parsing stuff. I think that person is also just a user as we are. 
If you check out also ch554 or ch552 stuff you will see that everywhere is used the header file but for ch554 or 2 of course. 
Also we do need the header as is as the special function registers are at that exact points. Hard to  descripe what i mean. Nobody could came up with a different file, something like that.

maikmerten, 2020-01-16:
Hmmm... I attached some pictures regarding file setup, commands and tool versions.

You're right, it makes perfect sense that the other programmer is a user just like us.

[image]

[image]

atc1441, 2020-01-16:
Thank you 

Did it all again right now and now it works to compile it :D

dont know what i did different this time...

But, the HEX file is different to the original one, and the WCHisptool is not accepting it. so there is something wrong.

Here is my windows compile.bat output: https://gist.github.com/atc1441/0365f070753d5edc416011926ce88b15

 By the way, du bist auch in deutschland ? ^^

maikmerten, 2020-01-16:
Bin auch in Deutschland. Hallo!

Wenn ich die Makefile mit der compile.bat vergleiche, dann fehlt die Entsprechung zu

`SDCC\bin\hex2bin -c %project_name%.hex`

Wenn ich das richtig sehe, dann soll hex2bin ja eine Hex-Datei wieder zu einer .bin-Datei machen... die haben wir aber doch eigentlich schon? Das Tool hex2bin ist bei sdcc standarmäßig wohl nicht dabei, aber ich kann es ja unter Wine ausführen:

```
maik@maik-desktop:~/SafeSync/programmierung/CH559sdccUSBHost$ wine SDCC/bin/hex2bin.exe -c CH559USB.hex 
hex2bin v1.0.1, Copyright (C) 1999 Jacques Pelletier
Lowest address = 00000000
Highest address = 00002977
```
Danach liegt eine identische .bin Datei wie vorher neu geschrieben herum (klar... bin zu hex und zurück sollte ja genau die gleichen Bytes liefern).

Da muss ich wohl noch ein bisschen herumgraben...

hexeguitar, 2020-01-26:
There is also another way to convert the ihex to bin using the _sdobjcopy_ file included in the sdcc toolchain:
`sdobjcopy -I ihex -O binary somefile.hex somefile.bin`
I believe on windows it would be:
`sdobjcopy.exe -I ihex -O binary somefile.hex somefile.bin`
Just tested on one of my projects: both output bin files, generated with sdobjcopy and hex2bin (with checksum flag) result in the same sha256 hash.