atc1441/CH559sdccUSBHost #5: Porting to CH554G

dmcnaugh, 2020-06-17:
Is there any experience or opinion with respect to porting this to the CH554G?

My understanding is that it has the same architecture but

- only 1 USB Host port
- and less memory

The advantage is that it comes in a SOIC-16 package and is therefore easier to hand solder.

atc1441, 2020-06-17:
Hi.

Porting should be possible, but look out for the differences, for example the gpio registers are not the same.

The ch554 was not that interesting as it is just 20cents cheaper than the more powerfull ch559

dmcnaugh, 2020-06-17:
Understood.
My motivation for the CH554G is only the ability to hand solder as it will be included in a kit where most other parts are through-hole. Many kit makers are put-off by SMD at the best of times.

Can I assume there is a header file for the CH554 similar to the CH559.h file in this repo?
Or do I have to hand craft that myself?
Thanks.

atc1441, 2020-06-17:
There is the CH554 file in my Other Repos about the CH55x Micros available,

By the way i made also an Arduino Board for the CH55x Micros, in there is also the CH554.h file 

https://twitter.com/atc1441/status/1272174545735221248

closed 2020-06-17 by dmcnaugh