atc1441/CH559sdccUSBHost #6: HID Keyboard - key codes with incorrectly set high bit set (0x80)

dmcnaugh, 2020-06-17:
I am testing with the pre-made board from MatzElectronics.
The only material change to the software is that the baud rate has been lowered to 400000.

For the keys 'A' to 'F6' I see the expected keycodes 0x04 to 0x3F
For all keys from 'F7' onwards I see keycodes from 0xC0 upwards
These should be 0x40 upwards according to the HID keyboard standard.
It appears that when the 0x40 bit is set then the 0x80 bit is being set incorrectly (noise?)

I have tried with a number of keyboards with the same result.
Around 1 in 50 times I will see the expected value.

I could mask the most significant bit, and this could work for most keyboards with less than 128 keys. But eventually this could be a problem.

Its this what other people experience?

dmcnaugh, 2020-08-12:
OK, finally got to investigate this.
Looks like it's a problem specific to the device on Tindie from MatzElectronics https://www.tindie.com/products/matzelectronics/ch559-usb-host-to-uart-bridge-module/

He has his own fork of the code that sets the baud rate to 400000 not 1000000 and this is what he ships flashed onto his device.

When I updated the firmware on that device to run with a baud rate of 1000000 the problem went away.
I have also made my own board based on the CH559T (SSOP-20) and it doesn't have this problem running at 1000000.

I didn't have to change the code for the CH559T, just flashed it and it worked. I chose the CH559T because:

1. I only need a single USB
2. I can hand solder a SSOP-20 package

atc1441, 2020-08-12:
Good to know what the error was. thank you for the update

closed 2020-11-14 by dmcnaugh