atc1441/CH559sdccUSBHost #9: Support USB-to-serial adapters

probonopd, 2020-10-03:
Would it be possible to add support for USB-to-serial adapters such as CDC, FTDI FT232xx, CH340, SiLabs CP210x, etc.?

This would be very useful e.g., to connect USB printers via USB to the ESP8266.

Example:
https://github.com/probonopd/WirelessPrinting/issues/157#issuecomment-703004262

atc1441, 2020-10-03:
Yes that can be added. There is even an example from WCH that does exactly that.

probonopd, 2020-10-03:
That would be _huge_. cc @TheAssassin

probonopd, 2021-04-10:
@atc1441 any chance you might get around to implementing this at some time?

atc1441, 2021-04-10:
Will most likely not add something in the near future, with the new ESP32 ( and even software usb host for the current esp32) and other devices that support usb host this chip is kinda obsolete. Also because it is not as cheap as the CH552

MirkoUgoliniDev, 2021-04-10:
Hi, could you  point  the example  in

> Yes that can be added. There is even an example from WCH that does exactly that.

could you  point  the  right code sample or attach it here?

Regards

Mirko

atc1441, 2021-04-10:
Here it should be http://www.wch.cn/products/CH559.html

MirkoUgoliniDev, 2021-04-10:
Aaron,

I'm sorry, but I couldn't find the code example. Could you please tell me which is the correct one?

A repository I found:
https://bitbucket.org/e_dragon/wch_ch55x/src/master/SDK/Official%20Examples/

probonopd, 2021-04-10:
@atc1441 

> Also because it is not as cheap as the CH552

Great point. The CH552 has

> Built-in USB controller and USB transceiver, support USB2.0 full speed and low speed **host** or device mode.

So I am asking myself, could a cheap USB host to serial converter be made, where you could attach e.g, a USB 3D printer on the one side, and get serial (that can be hooked up to an Arduino/ESP) on the other side.