atc1441/CH559sdccUSBHost #17: problem with full-speed USB device

tomtobback, 2021-07-13:
Hi, thanks a lot for sharing this project. it works well with low-speed USB devices (1.5Mbps), i have tested with a mouse and keyboard, but when i connect a high-speed USB device (12Mbps) like a barcode scanner, the firmware does not seem to deal with it correctly. in my case, the scanner, recognised as a keyboard, sends each char in the barcode as a single keystroke, and only the first few chars are correct, then some are incorrect, and some discarded; the results vary. i see in the firmware that the 2 ports are configured for full-speed but maybe there is a buffer or timing issue that causes this problem. any advice greatly appreciated.

DFL1973, 2023-05-05:
I second that. Full-speed gamepads seem to work incorrectly, but the low-speed ones are great with the firmware.

JoeHo8051, 2024-09-24:
This could be caused by the DMA marco address setting.
The original keil C compiler is little end style, and the author didn't change it to fit the sdcc compiler.

GrantMTG, 2024-09-25:
Are there any recent forks that use Keil C51?

JoeHo8051, 2024-09-28:
> Are there any recent forks that use Keil C51?

It seems no.