atc1441/CH559sdccUSBHost #20: Rebuilding and reloading the firmware

dbetz, 2022-07-01:
I've been trying to make some modifications to your firmware to support CDC and am running into a strange problem. I'm using Matt Matthew's module from Tindie and doing my work on a Raspberry Pi under Linux. I am able to use SDCC to rebuild the firmware and chflasher.py to flash it into the CH559 but the new firmware is not working. In order to try to figure out what might be going wrong I wrote a simple program that just toggles P0.2 and P0.3 so I can watch the signals on a logic analyzer but I don't see any activity when the program is running. Is there something I'm missing in my code? Really, my question is more whether there is something I'm missing about reprogramming the chip. This program uses your util.c code.

#include "CH559.h"
#include "util.h"

SBIT(LED02, 0x80, 2);
SBIT(LED03, 0x80, 3);

void main()
{
    initClock();
    PORT_CFG &= ~bP0_OC;
    PORT_CFG |= bP0_DRV;
    P0_DIR |= (1 << 2);
    P0_DIR |= (1 << 3);
    LED02 = 1;
    LED03 = 0;
    while(1)
    {
        if(!(P4_IN & (1 << 6)))
            runBootloader();
        LED02 = !LED02;
        LED03 = !LED03;
        delay(500);
    }
}

dbetz, 2022-07-02:
I think I made some progress on this. It seems that the hex2bin program I'm using may not be writing correct .bin files. I tried using sdobjcopy instead and was able to get the blink program above to work. I am still having trouble getting the CH559sdccUSBHost firmware working though.

dbetz, 2022-07-02:
I did some more work trying to track down this problem and it seems to have something to do with the program size. I have one version that works and chflasher says it 10912 bytes but when I add some code that isn't even called and the program size increases to 11398 it stops working. The load completes and verifies successfully but the program doesn't run. My understanding is that the CH559 has 64k of program memory so 11398 bytes should certainly fit. What could be going wrong?

dbetz, 2022-07-03:
One final update: It seems that the Linux tools I was using were causing the problem. When I moved the code over to a Windows machine and ran the compile.bat script I was able to build a .hex file that loaded and ran successfully. I'm not sure what is wrong with the Linux tools or my Makefile but they are apparently what was causing the problem. I've attached my Makefile in case anyone has time to look at it. I'd appreciate hearing about any problems you see in it.

[Makefile.zip](https://github.com/atc1441/CH559sdccUSBHost/files/9035140/Makefile.zip)
.

Emiluren, 2023-03-14:
Hm, I think I have run into a similar problem. If I flash the code directly from this repo on Linux it seemed like it crashed immediately. I made my own code based on this and got a bit further (I had to remove the initClock, not sure why) but I get strange results.

I connected a mouse to my ch559 and expected the following device descriptor:
`0x12 0x01 0x00 0x02 0x00 0x00 0x00 0x08 0xF0 0x03 0x4A 0x13 0x00 0x01 0x01 0x02 0x00 0x01` 

However, all I got was a sequence of repeating 0x12.

Then I found your thread so I tried using the bundled tools on Windows and it works! Even if I compile on Windows and just flash on Linux I get strange behaviour so it seems like the flashing tool is the problem. I will investigate some more

Emiluren, 2023-03-14:
Oops, I am stupid. I forgot to change `--xram-loc 0x0000` that I had from another example which overlaps with the USB RxBuffer :man_facepalming:. My project is [here](https://gitlab.com/Emiluren/ch559-usb-midi-host) in case anyone want to check it out